meshcore-dev/meshcore-ha #19: Feature Request: Explicit Paths

mikewren, 2025-04-06:
Unsure if the issue I’m experiencing with getting stats from newly added Repeaters many hops away is due to paths or not, but it would be nice to be able define the path hops that should be made to connect to a remote node.

Not sure how much is handled in firmware vs application layer, but there doesn’t appear to be any visibility into path hops in Home Assistant.

awolden, 2025-04-07:
I haven't experimented with paths much but you should be able to do this via cli commands.

I would recommended adding the [CLI card](https://github.com/awolden/meshcore-ha/issues/19) to a dashboard to make it easier.

After that you should be ale to do `change_path <ct> <path>`

Where CT is adv_name and path is the comma delimited path. The best way to track CLI commands right now is to enable debug logs while issuing the command.

awolden, 2025-06-10:
You can see the out path for a given node in the extra attributes for the sensor:

[image]

With 2.0.4 the command to change the path for a client changed

```yaml
action: meshcore.execute_command
data:
  command: change_contact_path 6b3fc7 ac03
```

Where `6b3fc7` is the pubkey prefix for the node you are targetting.

I have thought about how to handle the situation where paths changes, the client does it by resetting the path to the node on the client, which I am reticent to do with HA. I also might not want to hard-code that logic but expose it via automation. Since you can listen to raw MC events and issue commands it's possible to set-up automations that automatically set/reset paths for nodes, which might be the best path forward. It is definitely a problem to be solved though.

closed 2025-07-08 by awolden