meshcore-dev/meshcore-ha #21: Entity ID Changing When HA Reboots

mikewren, 2025-05-01:
I'm observing that the entity ID seems to be changing when Home Assistant reboots.

Sometimes it includes the physical(?) address of the node, sometimes it doesn't.

For example:

sensor.meshcore_ecabbc293b_direct_dups_six_acre_woods

vs. 

sensor.meshcore_direct_dups_six_acre_woods

I'm sure there's a predictable reason for this, but haven't been able to narrow it down yet - just noting the issue here for folks smarter than me to possibly identify root cause (and hopefully a solution). :)

awolden, 2025-05-01:
The HA entities are getting recreated by the integration on every reload of the platform. So this can especially happen with remotely monitored nodes if it wasn't able to get a successful ack on reload since it doesn't know the hash of the entity.

I don't think there is a silver bullet to solving this problem, but it can probably be handled better. I am in the process of a slow rewrite using the new meshcore py lib, which should provide some benefits in terms of stability and reliability, but it won't necessarily solved this problem outright. 

There are a couple other things we can do, such as fail entity creation if we can't reach the remote node to make the failed fetch more explicity, or use some sort of persistence work around to make HA aware of the node through restarts.

Hopefully that helps explain the behavior.

michaeljaddy, 2025-05-15:
> In the meantime, we could implement a few potential solutions:
> - Make entity creation fail explicitly when we can't reach remote nodes, making the failure more apparent
> - Develop a persistence mechanism that helps Home Assistant maintain awareness of nodes through restarts
> - Improve the hash storage/retrieval system to maintain consistent entity IDs

Thank you for all of your efforts so far with this integration!

Were you able to make any progress with any of the above workarounds? Having consistent entity IDs (whether they are populated or not) would be amazing.

awolden, 2025-05-21:
This should no longer be an issue with V2: https://github.com/awolden/meshcore-ha/releases/tag/v2.0.0

closed 2025-05-21 by awolden