meshcore-dev/meshcore-ha #23: Error: Sending Messages to Room Server

itsdoublearon, 2025-05-13:
It's been working fine for the past 24 hours, but then suddenly stopped....

Logger: custom_components.meshcore.services
Source: custom_components/meshcore/services.py:133
integration: MeshCore 

It's failing to send messages, even though it's reading them in the room just fine.

awolden, 2025-05-21:
Would be curious if this is still an issue in v2: https://github.com/awolden/meshcore-ha/releases/tag/v2.0.0

itsdoublearon, 2025-05-31:
I unplugged the node and connected to a different device, everything worked fine. Plugged it back into my HA instance and didn't work again. However did the set_time command you provided and it's working in HA again. I set it up as a automation to test that out.

awolden, 2025-05-31:
Thanks for the feedback. Might be good to have HA check the time on repeater failures and make sure its right on the node.

awolden, 2025-06-01:
I added automatic time syncing to help avoid this clock drift issue: https://github.com/awolden/meshcore-ha/releases/tag/v2.0.3

closed 2025-06-01 by awolden