meshcore-dev/meshcore-ha #26: Meshcore Forward to Push example doesn't work

orsib78, 2025-05-23:
Hello.

I tried to use the sample automation for "Meshcore Forward to Push", but it doesn't work.

Maybe the event itself not working correctly, because when I manually trigger the event (using Developer Tools), the result is an empty message, but when I send a test message to the Public channel it doesn't trigger the **meshcore_message** event at all.

Thanks,
Balazs

awolden, 2025-05-23:
Thanks for reporting, I found a couple issues with the example in `v2.0.0`. I just released `v2.0.1` and updated the example in the readme and it should work now.

orsib78, 2025-05-23:
Thanks. I’ll give it a try tomorrow.

orsib78, 2025-05-24:
> Thanks for reporting, I found a couple issues with the example in `v2.0.0`. I just released `v2.0.1` and updated the example in the readme and it should work now.

Hello. So, I tried your uplifted example automation... and now if I send a message from the node connected to HA, to the public channel, that triggers the HA example automation of yours... However if I send a message to the public channel using another node (eg. my t-echo) that doesn't trigger the HA automation... I double checked it and the HA node receives messages from the Public channel (the web application confirms it).

orsib78, 2025-05-25:
Hello. The notification forwarding is now working perfectly in 2.0.1. Thanks.

closed 2025-05-25 by orsib78