meshcore-dev/meshcore-ha #38: How can i send/receive Data with Meshcore-HA and put it into a variable?

Emmpunkt, 2025-07-06:
Hello,
i want to connect 2 HA Devices with each other over Meshcore and exchange some data.
Sending data is not the problem, but how do i receive the data and put it into variables?

What i´m trying to do:
I want to send some data from a solar powered remote hut to my home.
Something like battery state, produced energy, used energy. Twice a day might be enough. 
I already found out how to send a variable,, but i can´t find an example how to get the variable(s) back into HA-variables.

Thanks, Michael

awolden, 2025-07-06:
This is really more of a general HA question, but you can create an automation that fires off incoming message and handle it accordingly:

```yaml
# First, create input_text helpers to store the received data
# Add this to your configuration.yaml:
input_text:
  received_message:
    name: "Last Received Message"
    max: 255
  
  sender_node:
    name: "Message Sender"
    max: 100

input_number:
  battery_level:
    name: "Remote Battery Level"
    min: 0
    max: 100
    unit_of_measurement: "%"
  
  energy_produced:
    name: "Energy Produced"
    min: 0
    max: 1000
    unit_of_measurement: "kWh"

# Then create the automation:
automation:
  - alias: "Receive MeshCore Message and Store Variables"
    description: "Process incoming MeshCore messages and extract data"
    trigger:
      - platform: event
        event_type: meshcore_message
        event_data:
          message_type: direct  # or channel depending on your setup
    
    condition:
      # Optional: filter by specific sender
      - condition: template
        value_template: "{{ trigger.event.data.from_node_id == 'RemoteHut' }}"
    
    action:
      # Store the raw message
      - service: input_text.set_value
        target:
          entity_id: input_text.received_message
        data:
          value: "{{ trigger.event.data.message }}"
      
      # Store the sender info
      - service: input_text.set_value
        target:
          entity_id: input_text.sender_node
        data:
          value: "{{ trigger.event.data.from_node_id }}"
      
      # Parse structured data (assuming format like "battery:85,energy_produced:12.5")
      - service: input_number.set_value
        target:
          entity_id: input_number.battery_level
        data:
          value: >
            {% set msg = trigger.event.data.message %}
            {% set battery = msg | regex_findall('battery:([0-9.]+)') %}
            {{ battery[0] if battery else 0 }}
      
      - service: input_number.set_value
        target:
          entity_id: input_number.energy_produced
        data:
          value: >
            {% set msg = trigger.event.data.message %}
            {% set energy = msg | regex_findall('energy_produced:([0-9.]+)') %}
            {{ energy[0] if energy else 0 }}
      
      # Optional: Log the received data
      - service: system_log.write
        data:
          message: "Received MeshCore data: {{ trigger.event.data.message }}"
          level: info
```

Emmpunkt, 2025-07-07:
Wow, i did not expect a "ready to use" answer. 
Many thanks!!
Maybe you should put this in the Wiki or the read.me. I´m shure this is useful for others too.

Yes, you are right, this is a HA related question. The problem is how to "articulate" my question to google or a forum. I do not really know what i should ask for if i had to explain the whole "meshcore thing" first.
So its easyier to ask for help in the associated github/forum/whatever....
And YAML is a thing that i do not really understand.
And I´m german, english makes it even harder. ;)

I got the automation running, but i had to deactivate the "message-sender" part.

`  condition:
      # Optional: filter by specific sender
      - condition: template
        value_template: "{{ trigger.event.data.from_node_id == 'RemoteHut' }}"
`
It did also not work for the other examples in the readme, seems to be always empty.

Many thanks Michael

closed 2025-07-28 by awolden