meshcore-dev/meshcore-ha #40: How to configure update intervals?

mrlmuk, 2025-07-22:
In the setup information it mentions...

Configure update intervals for different data types:
Messages interval: How often to poll for new messages (default: 10 seconds)
Device info interval: How often to update device statistics (default: 60 seconds)
Repeater update interval: How often to poll repeater nodes (default: 300 seconds)

I found some settings in /homeassistant/custom_components/meshcore/const.py but they don't obviously match these three mentioned above.

Should this be being updated via the files or have I missed something in the main HA GUI?

schnobbc, 2025-07-25:
I also wonder how to change this after the node is added

mrlmuk, 2025-07-27:
The reason for wanting to change this I have no need for 10 second updates of message. The reliability issues I am having may be made far worse by frequent updates failing and they may go away entirely if this is increased to several minutes. Also this must be affecting battery life on the ESP models. I don't want and in some cases cannot to have things permanently plugged into the mains supply.

awolden, 2025-07-28:
Ya these were changed in refactor the newest version of the SDK. I can look at exposing more configuration optins in future releases

mrlmuk, 2025-08-11:
I'm convinced this is badly affecting battery usage. Im using esp32 S3 which isn't the most efficient but when the disconnects happen it seems the battery lasts a lot longer.  I will be moving to nrf for the repeater soon anyway but the main node will need to stay WiFi as I cannot get BT working and don't want no pin mods.

awolden, 2025-09-07:
More configuration options have been added for this. Namely around contact fetching which is the vast majority of the data over the wire to HA