meshcore-dev/meshcore-ha #44: V2.0.9 Does not connect to radios over ble

Delphinus44, 2025-07-29:
Installed the last version and integration fail to connect over BLE, tested with 2 different radios.

Message:  Failed to connect

awolden, 2025-07-29:
Was it working before? How are you connecting the BLE nodes?

Delphinus44, 2025-07-29:
Never worked, install it yesterday and never manage to connect, it can see the radios but fail to connect.
Have 4 bluetooth devices connected and even stop the instance that uses the bluetooth dongle but even without other connections got the same error. 
Do I need to edit a file to connect?
Thanks

awolden, 2025-07-29:
A lot of depends on how the connection happens and how you are pairing it with HA. If you are using some like a ble proxy using a esp board I dont think there is a way to get it paired with a pin. The integration itself handles doesn't handle setting up a BLE connection to the device, it just lets you select an existing BLE connection that HA detects. The management of BLE connections isn't really in the scope of the integration