meshcore-dev/meshcore-ha #63: Connection Issue 1.7.1<

itsdoublearon, 2025-09-27:
I tried updating my Heltec V3 to 1.8.1 but would not connect to the meshcore-ha, even did a full reboot of Home Assistant OS and still nothing. Tried different firmware on the radio, but had to rollback to 1.7.1 and it was read again in Home Assistant.

awolden, 2025-09-28:
Can you try connecting with the [meshcore cli](https://github.com/meshcore-dev/meshcore-cli/tree/main) that will give some indication if its HA related or meshcore sdk related

itsdoublearon, 2025-09-30:
Just tried with the latest 1.9.1 firmware and still not connection with HA, but cli worked.

awolden, 2025-09-30:
I flashed a heltec v3 from the web flasher and it was not able to connect. I flashed from the same node from source on the latest dev branch and it works fine. Not sure if there is an issue with the web flasher or if a bug as been fixed in dev

itsdoublearon, 2025-10-02:
My Heltec V3 is connect via USB to my computer running HA

itsdoublearon, 2025-10-04:
Just updated to MeshCore-HA Version 2.1.7 - Same issue with my Heltec V3 (USB Companion) not connecting for firmware higher than 1.7.1

kaczoralex-cell, 2025-10-04:
I currently cant connect a heltec v3 on 1.9 to HA via USB as well

awolden, 2025-10-04:
Can you guys try the nightly build?

https://files.brazio.org/meshcore/nightly

lnitram, 2025-11-18:
I updated my heltec V2 (yes, old but did the job so far) to the new 1.10. firmware. Since then I cannot connect it to HA anymore. 

- It worked fine on my Ubuntu Laptop using the "Web Client by Liam Cottle - Chrome browser and via iOS Bluefy app"
- I can see the device on HA (Running on RPI 4) in the hardware list (ttyUSB1, /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0)
- When I add it to HA and select USB as a connection and put /dev/ttyUSB1 as a device, I get " Failed to connect" after a few seconds
- I remember I had similiar issues when I went to 1.9 and I think a downgrade back to 1.8 solved it. (not 100% sure anymore)